modelica-3rdparty/Modelica_DeviceDrivers #284: Release v1.7.0 not backwards compatible

bernhard-thiele, 2019-04-02:
Commit 1b03b84ed0573c46c8f36c5a864ccfe0ed0daeeb added a dependency to an MSL icon which was newly introduced in MSL v3.2.3. As a result MDD v1.7.0 unintentionally will not work if used with a previous MSL version. We should probably aim for a bug fix release v1.7.1 which establishes backwards compatibility.

closed 2019-04-02 by bernhard-thiele

tbeu, 2019-04-02:
So basically this means, we only increase the version of the uses annotation (as e4c854a47c961a57634d47171c60d84968b1147c) did, to please the Dymola auto-conversion, but nost not actually update library code to utilize new features. Sigh. If we only had [impact](https://impact.github.io/#/).

bernhard-thiele, 2019-04-03:
Unfortunately, I don't know a better way using the current version handling facilities. This `uses(...)` annotation should be enhanced, but I suppose that would mean to agree on a common versioning scheme like [SemVer](https://semver.org/) and support it in a standardized fashion.

bernhard-thiele, 2019-04-03:
I've made a draft release for v1.7.1 and planned to publish it this Friday, 2019-04-05. This can be moved to a later date if there pop up some other issues which should be solved.

commit 86020317 (2019-04-02): Revert "Utilize Modelica.Icons.FunctionsPackage"

This reverts commit 1b03b84ed0573c46c8f36c5a864ccfe0ed0daeeb.

Fixes #284.