modelica-3rdparty/Modelica_DeviceDrivers #292: Getting #error during Modelica to EmbeddedC code generation

Hematies, 2019-07-31:
Hi. I was following the steps related to the code generation and compilation of Modelica models using DeviceDrivers for OpenModelica when I tried using an example model such as LogicalNetwork (Modelica.Blocks.Examples.LogicalNetwork1). All seemed to be successful until I realized in the generated code that lines of this kind have been written:
```
if (
  #error "[CodegenEmbeddedC.tpl:474:14-474:14] daeExpBinary: Not supporting operator? size(table, 1) == 0"
  )
```
A scarier example:
```
#error "[CodegenEmbeddedC.tpl:364:14-364:14] Unsupported statement: when $whenCondition1 then
    if table1.index < table1.n then
      table1.index := 1 + table1.index;
      table1.nextTime := table1.table[table1.index];
      table1.y := not table1.y;
    elseif table1.index == table1.n then
      table1.index := 1 + table1.index;
      table1.y := not table1.y;
    end if;
  end when;"
```
(I don't even recognize this syntax, which is quite similar to PL/SQL)

Do I have to worry about this code? Are these errors to be considered? In that case, how do I fix them?

Thanks in advance!!

[LogicalNetwork_main.zip](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/files/3450810/LogicalNetwork_main.zip)

sjoelund, 2019-07-31:
Adding support for `==` is simple (https://github.com/OpenModelica/OpenModelica/pull/362)

But there is no support for dynamic sizes of arrays or arrays in functions; the size operator is also expected to have been evaluated and would cause problems.

There is no support in that code generator for when-statements; using if-statements as a workaround is possible by changing some libraries.

Using things like the table implementation from the standard library is not possible due to using all of these language features.

Hematies, 2019-07-31:
Given the fact I have just started to use OpenModelica and DeviceDrivers (I am a noob), my next question is: what is the range of applicability of EmbeddedC within the Modelica system libraries then? Would you recommend it for extended use or just take it with caution? Would the unsupported features be easily corrected by a user like me (just like in the case of OpenModelica/OpenModelica#362)?  

Thank you very much for your quick response!

sjoelund, 2019-07-31:
You will only be able to use limited parts of the Modelica Standard Library with the EmbeddedC code generator - it was created for very limited systems such as Atmel microcontrollers which don't have a floating-point unit, a file system, or the memory allocations needed for many of numerical algorithms that are standard for Modelica tools. Most Modelica libraries assume you have a powerful PC.

If what you have is a more powerful device, you are often better off with source-code FMUs (cross-)compiled for that platform.

closed 2019-08-05 by bernhard-thiele