modelica-3rdparty/Modelica_DeviceDrivers #294: Spacemouse not working under Windows 10

bernhard-thiele, 2019-09-03:
When using the MDD Block for the 3DConnexion SpaceMouse on Windows 10, the block doesn't get the inputs from the SpaceMouse (it just returns zeros).

bernhard-thiele, 2019-09-03:
The problem is that the length of the list of scanned potentially relevant USB devices is limited by a too low hard-coded value:

https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/blob/3f6342e6b6d0bb84bccac41ca7682a9eaba6393d/Modelica_DeviceDrivers/Resources/src/SpaceMouse/MDDSpaceMouse.c#L23

At least for some of our Windows 10 systems the list needs to be longer, otherwise the SpaceMouse device will not appear in the list.

bernhard-thiele, 2019-09-03:
Still need to provide the updated binaries

closed 2019-09-03 by bernhard-thiele

tbeu, 2019-09-03:
This requires new binaries MDDSpaceMouse.[lib, dll, a], right?

bernhard-thiele, 2019-09-03:
Yes, I just pushed a VS 2010 generated .lib and .dll using an old Windows Driver Kit (WDK) that I still have at home (Version 7.1.0) for backwards compatibility. Can't check if it works here (no SpaceMouse at home), but will check it tomorrow in my office. It is not needed to update the Linux binaries, since the changes are only for Windows.

tbeu, 2019-09-04:
The updated MDDSpaceMouse.dll (both win32 and win64) does not export MDD_spacemouseGetData.

bernhard-thiele, 2019-09-04:
Thanks for spotting this error. Should be fixed with f6f10fc.

tbeu, 2019-09-05:
Is this issue really specific for Win10?

bernhard-thiele, 2019-09-05:
No, not really specific to Win10. The issue with the previous code was that it limited the number of HID devices that were checked for being a SpaceMouse at a (too low) max value. For some reason that seems to be (usually?) no problem in Windows 7, but in Windows 10 the SpaceMouse appeared at an index that was not checked anymore. I've simply increased this max value from 10 to 50. This seems to be enough (though maybe there exists configurations there this is too low). It's a quick fix that got it working. It's probably not a super solid solution. I don't really want to invest too much time trying to understand the details of this Windows driver API, but I'm of course open if there are suggestions for a better solution.

commit e62cc377 (2019-09-03): Increase MAX_DEVICES to 50 (Fixes #294)