modelica-3rdparty/Modelica_DeviceDrivers #300: Send / receive on a serial port within the same program

max-privato, 2019-11-12:
I would like to use the serial port of Device drivers to send and receive data from/to the same modelica program.
I did this successfully using two serial ports of my PC, one to send the other to receive.

I can survive this, but I wonder whether there is a way to use the same port to both send and receive data.  This is a typical easy task done with serial ports, and therefore it seems weird to me that DeviceDrivers library does not allow this.

It this a library limitation I just could not find the right way to do so?

bernhard-thiele, 2019-12-04:
@max-privato I think we found a solution for your intended application by direct communication (using UDP instead of the serial port), but since this issue might be interesting to other people as well, I just write a comment here.

You are right, it is currently not supported when using the block interface, i.e., blocks `SerialPortReceive` and `SerialPortSend`, which create independent external objects. It's a long time since I worked on the code, but one could probably make a "custom" block which works bidirectional by building directly on the function layer and create a single external object (configured as "receiver") and use that object also as argument in the `sendToSerial()` function.

In order to have separate blocks for sending and receiving, one could share the external object, similarly as I have done it in the new TCPIP server (see example `Blocks.Examples.TestSerialPackager_TCPIPServer`). There, an `inner` configuration block holds the external object, which makes it accessible by other read and writes blocks. I currently don't plan to add something similar for the serial port, but it would be nice to have

closed 2019-12-04 by bernhard-thiele

max-privato, 2019-12-08:
> @max-privato I think we found a solution for your intended application by direct communication (using UDP instead of the serial port), 
yes, thank you for your help also with UDP

Thank you also for giving hints on how to improve the serial communication. Maybe not me (I don't have the necessary knowledge) but others or even some colleagues of mine in the future can use your hints.