modelica-3rdparty/Modelica_DeviceDrivers #301: MQTT not working with Visual Studio 2017

PMehrfeld, 2019-11-22:
# Problem
I tested the example `Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_MQTT` on two machines with the following configuration:

- Windows 10
- Dymola 2020
- **Visual Studio (VS) 2017**

The example crashed with following error in the translation tab:
[image]

# Step GCC
As a conclusion I tried with GCC compiler (32-bit GCC version 4.8.1, 64-bit GCC version 4.9.2). 

I received the following error:
```
Compiler message:

Compiling and linking the model (GCC). 

C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5185): undefined reference to `MQTTClient_freeMessage'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5190): undefined reference to `MQTTClient_free'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5488): undefined reference to `MQTTClient_setTraceLevel'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5494): undefined reference to `MQTTClient_setTraceCallback'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x54a2): undefined reference to `MQTTClient_setTraceCallback'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5506): undefined reference to `MQTTClient_create'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x55bb): undefined reference to `MQTTClient_setCallbacks'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x55e9): undefined reference to `MQTTClient_connect'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5619): undefined reference to `MQTTClient_subscribe'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5632): undefined reference to `MQTTClient_strerror'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x56a0): undefined reference to `MQTTClient_destroy'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5760): undefined reference to `MQTTClient_strerror'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x582c): undefined reference to `MQTTClient_unsubscribe'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5894): undefined reference to `MQTTClient_disconnect'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x58a1): undefined reference to `MQTTClient_destroy'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5978): undefined reference to `MQTTClient_publishMessage'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x59ab): undefined reference to `MQTTClient_waitForCompletion'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x59c1): undefined reference to `MQTTClient_strerror'
C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o:dsmodel.c:(.text+0x5bdd): undefined reference to `MQTTClient_getVersionInfo'
d:/06_programme/mingw32_gcc_v4_8_1/bin/../lib/gcc/i686-w64-mingw32/4.8.1/../../../../i686-w64-mingw32/bin/ld.exe: C:\Users\<user>\AppData\Local\Temp\ccqQL2un.o: bad reloc address 0x0 in section `.data'
collect2.exe: error: ld returned 1 exit status
```

# Step VS 2015
This works.

# Question
Is it a general issue that VS 2017 seems not to be fully supported?  
Is it planned to be supported for MQTT?

bernhard-thiele, 2019-11-22:
Thanks for the report. It's supposed to work with VS 2017. I just checked the current master on my Windows 7 machine using Dymola 2020 and VS 2017 as compiler, and it worked.

The problem that probably hit you there is that the remote MQTT test server ("test.mosquitto.org") which is used in the example is not always responsive. Please try several times whether the problem persits.

The example should work with GCC under Linux, but I've not tested it with MinGW under Windows and it is possible that the needed binary dependencies (DLLs) which are usually generated by @beutlich using Visual Studio don't work with MinGW.

beutlich, 2019-11-23:
Cannot reproduce. Works as expected in my environment, being

* Windows 10
* Dymola 2020 64-bit
* Visual Studio (VS) 2017
* M_DD f6f10fc1d7e3404b27a0f11b75d56c0467fe93e3

Please check your firewall settings. I used to have issues with disabled ports on my former business machine. But in that case the error message was rather explicit, that no connection could be created.

PMehrfeld, 2019-11-25:
Thanks both @bernhard-thiele and @beutlich: After additionally installing VS 2015 and switching back to VS 2017 in the Dymola compiler settings, it works also with VS 2017...  
Maybe, the VS 2017 installer, distributed by our IT division, does not install all required components of VS, which are delivered with the older VS 2015 installer.
Sorry for bothering you, but since the simulation stopped without any error indication, I first addressed the issue tracker of this project.

closed 2019-11-25 by PMehrfeld