modelica-3rdparty/Modelica_DeviceDrivers #306: MQTT Linux library dependencies should be recompiled with -fpic

bernhard-thiele, 2020-03-24:
This is an issue reported to me which popped up when trying to translate an FMU using Dymola:
 
```
translateModelFMU("Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_MQTT annotation(experiment(StopTime=5, __Dymola_Algorithm=\"Dassl\"))", false, "", "2", "all", false, 0);
= ""
 
…
/usr/lib64/gcc/x86_64-suse-linux/5/../../../../x86_64-suse-linux/bin/ld: /home/zpf/dymoladists/dymolaR423/Modelica/Library/Modelica_DeviceDrivers 1.8.1/Resources/Library/linux64/libssl.a(s23_clnt.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with –fPIC
…
/usr/lib64/gcc/x86_64-suse-linux/5/../../../../x86_64-suse-linux/bin/ld: /home/zpf/dymoladists/dymolaR423/Modelica/Library/Modelica_DeviceDrivers 1.8.1/Resources/Library/linux64/libcrypto.a(mem_dbg.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with –fPIC
…
 ```

So to work with FMI, the libraries (only listed a few above) under Resources/Library/linux64 need to be compiled with the -fPIC flag.

beutlich, 2020-03-24:
Right, can do. Any deadlines?

bernhard-thiele, 2020-03-24:
No, would be good to have it in the next release, but there is no milestone planned, yet. If it is done earlier I would create an intermediate bug fix release, so that it has a chance to be shipped with the next Dymola release.

beutlich, 2020-03-24:
This applies to libcrypto.a and libssl.a. Also to libpaho-mqtt3cs.a?

sjoelund, 2020-03-24:
That one should be fine.
```
$ rm *.o
$ ar x ~/Downloads/libpaho-mqtt3cs.a
$ clang -o test.so -shared *.o
$ ar x ~/Downloads/libssl.a
$ clang -o test.so -shared *.o
/usr/bin/ld: bio_ssl.o: relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: d1_both.o: relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
```

tbeu, 2020-03-24:
> Also to libpaho-mqtt3cs.a?

libpaho-mqtt3cs.a is built with -fPiC already.

https://github.com/tbeu/paho.mqtt.c/blob/6c8f6346e442f2ed35e13b5b7db06196b42eebfb/.travis.yml#L56-L57

tbeu, 2020-03-24:
@bernhard-thiele In the spirit of TDD can you please add a unit test, where we compile a shared library (instead of an executable).

bernhard-thiele, 2020-03-25:
Yes, will do.

PR #307 (2020-03-25): Update OpenSSL to 1.0.2u
Linux libraries have been rebuilt with -fPIC to fix #306.
closed 2020-03-25 by bernhard-thiele