modelica-3rdparty/Modelica_DeviceDrivers #312: Packager buffer overflow for vectors sized above 3

pierrebarroca, 2020-04-23:
When I try to add a vector with size above 3 the package buffer overflows.

Below the error log on dymola

dymosim started
... "dsin.txt" loading (dymosim input file)
MDDUDPSocket.h: Opened socket for sending.
MDDUDPSocket.h: Waiting for data on port 10002.
The following error was detected at time: 0
MDDSerialPackager.h: MDD_SerialPackagerAddDoubleAsFloat failed. Buffer overflow.

The stack of functions is:
Modelica_DeviceDrivers.Packaging.SerialPackager_.addRealAsFloat
Modelica_DeviceDrivers.Blocks.Packaging.SerialPackager.Internal.DummyFunctions.addRealAsFloat

tbeu, 2020-04-23:
Does it help if you increase the buffer size or utilize the automatic buffer size propagation feature in packager block and sender/receiver block?

[image]

[image]

pierrebarroca, 2020-04-23:
I tried with autoBufferSize checked and also changing userBufferSize.
You can trigger the same error with the TestSerialPackager_UDP example in the library by changing the size of the vector inputs for addReal and addFloat blocks.

The addReal block raises error for vectors with size above 3 and the addFloat for vector sizes above 2.

tbeu, 2020-04-23:
Did you adapt the settings in all three blocks: sender, receiver and packager block?

pierrebarroca, 2020-04-23:
I solved the problem.
Indeed, I was adapting all the settings settings only on the sender and receiver and not in the packager block.
Thank you @tbeu !

closed 2020-04-24 by tbeu