modelica-3rdparty/Modelica_DeviceDrivers #319: Make Release v2.0.0

bernhard-thiele, 2020-06-04:
The library has been migrated to MSL 4 and I've made tests with Dymola and OpenModelica (I restricted myself to run the examples which don't need special hardware).

- Dymola 2021 (Linux and Windows): Works.
- OpenModelica 1.16.0~dev-420-gc007a39 (Linux): Blocks.Examples works, ClockedBlocks.Examples using SerialPackager fails. However, the failing examples also fail in MDD v1.8.2 with MSL 3, so I think there is no degration because of the conversion to MSL 4.

Therefore, I think we can start  to update release notes etc. and prepare release v2.0.0. 

@tbeu @HansOlsson @sjoelund Anything else which should be considered?

beutlich, 2020-06-05:
I will provide another update of paho.mqtt.c soon.

bernhard-thiele, 2020-06-08:
Done (https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/releases/tag/v2.0.0).

closed 2020-06-08 by bernhard-thiele