modelica-3rdparty/Modelica_DeviceDrivers #326: Modelica_deviceDrivers is not compatible with FMUs

JyothiMatam6, 2020-10-30:
A model with  both Modelica_DeviceDrivers and FMU does not compile due to multiple definitions of several Winsock data types.
In the windows kit its clearly mentioned that "#error Do not include winsock.h and ws2def.h in the same module. Instead include only winsock2.h." 
Modelica_DeviceDrivers  is using windows.h and all the FMUs include a header file that includes windows.h which includes winsock.h. So there is a conflict.

please find the attached log.
[dslog.txt](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/files/5465021/dslog.txt)

beutlich, 2020-10-31:
Strange, we had/fixed a similar issue in #11. Can you please attach a [SSCCE](http://sscce.org/) that reproduces that compilation error?

beutlich, 2020-11-01:
Additionaly, can you please change line 245 of E:/Dymola2021_Installation/bin/build.bat from

```
set CL=%CL% /nologo /W3 %X64Define% /D"WIN32" /D"_CONSOLE" /D"DYMOSIM" %DYNSUNDIALSCL% /Oy /Ob1 /Oi /MT /D"_CRT_SECURE_NO_DEPRECATE" %DYMOLACL%
```

to

```
set CL=%CL% /showIncludes /nologo /W3 %X64Define% /D"WIN32" /D"_CONSOLE" /D"DYMOSIM" %DYNSUNDIALSCL% /Oy /Ob1 /Oi /MT /D"_CRT_SECURE_NO_DEPRECATE" %DYMOLACL%
```

and provide the enriched dslog.txt / buildlog.txt files.

beutlich, 2020-11-01:
Ah, I can reproduce now. The missing piece of information was, that the FMU also must be exported by Dymola since the compilation error does not occur for other FMUs.

This is an issue of the Dymola 2021 FMI sources. My recommended workaround (for you to continue) is to replace

```c
#include "windows.h"
```

by

```c
#if !defined(WIN32_LEAN_AND_MEAN)
#define WIN32_LEAN_AND_MEAN
#endif
#include "windows.h"
```

in both E:\Dymola2021_InstallationSource\FMI\fmiImport.h and E:\Dymola2021_InstallationSource\FMI\fmi2Import.h

@d94pn @HansOlsson FYI

KarlWernersson, 2020-11-06:
@beutlich thanks for the analysis 
Fixed for Dymola 2022

If you need for previous version you can find the headers in your "dymola install dirctory" \Source\FMI

backup your files fmiImport.h and fmi2Import.h
Then apply the fix recommended by @beutlich 

Alternatively contact Dymola support and we can provide a fix for you.

closed 2021-01-12 by bernhard-thiele