modelica-3rdparty/Modelica_DeviceDrivers #333: Joystick and keyboard not working in Fedora 33/ OMEdit/ various versions of device drivers.

linuxguy123, 2021-01-28:
Trying to run Modelica_DeviceDrivers->Blocks->Examples->TestInputJoystick in OMEdit.

Tried with several versions of omlib-modelica-devicedrivers installed from from https://build.openmodelica.org/rpm/fc33/ using dnf install <package>

Tested 1.7.1, 2 <- wouldn't install due to package conflict.
Tested 1.8.0,1,2 <- installed and ran in OMEdit.  No output though.
Tested 2.0.0 <- not backwards compatible, I know... tried it anyway.

Other packages installed on the computer:

# dnf list \*modelica\*
Last metadata expiration check: 1:57:33 ago on Wed 27 Jan 2021 04:29:19 PM MST.
Installed Packages
omlib-modelica.noarch                                                            20190520_223748~git~master_om1-1                                       @openmodelica
omlib-modelica-3.2.3.noarch                                                      3.2.3_6_g21327c6_om2-1                                                 @openmodelica
omlib-modelica-devicedrivers-1.8.2.noarch                                        1.8.2-1                                                                @openmodelica
omlib-modelica-synchronous-0.93.0.noarch                                         0.93.0_8_g5aa86a2-1                                                    @openmodelica
omlib-modelicareference.noarch                                                   20210119~124042~git~OM~maint~3.2.3-1                                   @openmodelica
omlib-modelicaservices-3.2.3.noarch                                              3.2.3_6_g21327c6_om1-1                                                 @openmodelica
omlib-modelicaservices-trunk.noarch                                              20200616~133533~git~OM~trunk-1                                         @openmodelica
openmodelica-1.16.x86_64                                                         1.16.2-2.fc33               

uname -a
Linux bigboy 5.10.9-201.fc33.x86_64 #1 SMP Wed Jan 20 16:56:23 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Output from simulation window:

/tmp/OpenModelica_me/OMEdit/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick -port=35615 -logFormat=xmltcp -override=startTime=0,stopTime=20,stepSize=0.04,tolerance=1e-06,solver=dassl,outputFormat=mat,variableFilter=.* -r=/tmp/OpenModelica_me/OMEdit/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick_res.mat -w -lv=LOG_STATS -inputPath=/tmp/OpenModelica_me/OMEdit/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick -outputPath=/tmp/OpenModelica_me/OMEdit/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick
Found joystick: FrSky RadioMas TX16S Joystick
        axis: 7, buttons: 24

Trying to set ProcessPriority: 0.

ProcessPriority set to 0=nice(0) "normal".

The initialization finished successfully without homotopy method.
### STATISTICS ###
The simulation finished successfully.

Symptoms:
- when I run the simulation in OMEdit, there is no change in the output variables shown in the Variables Browser window.   
- when I view the joystick device in KDE->System Settings, the axis values change on Axes 1 through 6, as they should.
- I have a piece of test code for another application that uses the /linux/joystick device driver and the joystick output works there too.  
- the keyboard test doesn't appear to work either.

I am a OMEdit newbie.   Am I missing something ?

Let me know if you need any other information.  I'm willing to debug code from my end if I can obtain a bit of direction.

Keep up the good work !  I thought I was going to have to write my own joystick driver... happy to have one in the library.

bernhard-thiele, 2021-01-28:
Thanks for the report. In the moment I have no joystick to test it at home, but I'll try to get my hands on one.

What OpenModelica version are you using?

It would be interesting whether starting the simulation binary (`/tmp/OpenModelica_me/OMEdit/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick`) without OMEdit on a console gives the same result. I think at least for the keyboard it was an issue that it wouldn't work in OMEdit, but works on the console.

bernhard-thiele, 2021-01-29:
I've tested MDD v1.8.2 with OMEdit/OpenModelica installed from the nightly build repository (OpenModelica 1.17.0~dev-342-gcbdefcd)  on Ubuntu 18.04.5 LTS connected to a Sony Playstation(R)3 Controller (`/dev/input/js0`). It worked.

Since it finds your joystick it should also be the correct `ID` in the `JoystickInput` block parameters. So I've no idea what goes wrong. I've seen you are using `openmodelica-1.16.x86_64` and I don't see why the nightly build should work better, but using a nightly build OpenModelica is one thing which one still could try.

linuxguy123, 2021-01-30:
Thanks for the reply and checking things out.

I'll install the nightly build and report back.

linuxguy123, 2021-01-30:
Newbie question... how do I run a block outside of OMEdit and observe the output ?

linuxguy123, 2021-01-30:
To run outside of OMEdit, I am running:

./tmp/OpenModelica_me/OMEdit/Modelica_DeviceDrivers.Blocks.Examples.TestInputJoystick/TestInputJoystick

How do I observe the output ?  Will it print to stdout ?

bernhard-thiele, 2021-01-31:
Hm, just realized that you replied. OMC generates C-code and builds a simulation binary. Executing this binary creates a *.mat result file which is automatically loaded into OMEdit after the simulation finished. If you run from the command line you, it will still create a *.mat-file which you can import into OMEdit (File->Open Result File). 

> Do I need to connect anything to the output of the joyStickInput block ? I'm not. I'm enabling the axes and buttons output and watching them on the graph.

Nothing needs to be enabled. Just simulate and move the stick. After the simulation is finished plot the signals in the `joystickInput.axes` array and check whether you see some values have changed.

> How do I observe the output ? Will it print to stdout ?

I'm not sure if you see all the output on stdout. Might be some goes directly into the logfiles. Best watch stdout and the *.log file.

> If I can find the MDT plug in for Eclipse, I'll step through the code and see what I find.

MDT might not behave as you would expect. It is mostly a tool for developing the OM Compiler (OMC) which is written in MetaModelica. You can also use it for editing conventional Modelica code, but the features are not as good as in OMEdit. Particularly, step-by-step debugging only works for algorithmic code and and not for simulation models (afaik). There are different options for debugging Modelica models in OMEdit, but for me it is usually simply to simulate, check plots and usage of `Modelica.Utilities.Streams.print()`.

linuxguy123, 2021-02-02:
Got it !   Thanks for the reply.   I'll dig into this tonight.

bernhard-thiele, 2021-03-04:
Closing the issue since there was no activity for some time

closed 2021-03-04 by bernhard-thiele