modelica-3rdparty/Modelica_DeviceDrivers #334: MQTT : Subscribe to multiple topics with one client

Maghnie, 2021-02-22:
Hello, are there plans to extend the MQTT client library to support the `subscribeMany` function from [paho.mqtt.c](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/blob/master/Modelica_DeviceDrivers/Resources/thirdParty/paho.mqtt.c/MQTTClient.h#L1113) ?

Currently, we (@PMehrfeld) subscribe to multiple topics using multiple MQTT clients. However we prefer setting up just one client during the simulation.

Either way, thanks for the helpful library. 🙂

PMehrfeld, 2021-05-19:
@bernhard-thiele or @tbeu: Sorry for pushing this issue, but do you have any opinions regarding the request in this issue? Thanks in advance!

bernhard-thiele, 2021-05-20:
Hi @Maghnie  and @PMehrfeld, I just discussed the issue with @tbeu who implemented and maintains the MQTT feature (so far I don't know very much about MQTT). It seems like an interesting feature, but somebody would need to jump in for implementing it. Unfortunately, there are no spare resources for us to work at this. So I'll close the issue for now since there is nobody who could work at it. Otherwise, it is a good feature request and it would be nice to have it.

closed 2021-05-20 by bernhard-thiele