modelica-3rdparty/Modelica_DeviceDrivers #336: Transfer a large amount of data from Dymola to MATLAB via UDP

Satinelamp, 2021-04-02:
I am considering using the Modelic_DeviceDrivers library to transfer data via UDP, but it turns out I have to connect every variable via a component, and I have a few thousands of variables, it would take too much effort to connect them.<Br>
How could I transfer a large amount of data from Dymola to MATLAB efficiently via UDP?

bernhard-thiele, 2021-04-06:
Maybe you could mechanize it by using the underlying Modelica functions directly, instead of using the drag'n drop blocks.
Though I'm not sure whether it is a good idea using this UDP interface if you have so many variables. It could work, could be even rather efficient, but it simply stuffs everything sequentially in one UDP message. It's rather easy to introduce offset bugs if message construction and deconstruction don't fit exactly and it is potentially hard to debug if you have so many variables.

closed 2021-04-06 by bernhard-thiele

Satinelamp, 2021-04-06:
Except for UDP, is there any other method that allows massive data exchange via the internet efficiently?

bernhard-thiele, 2021-04-06:
UDP may drop packets, it is particularly for time-sensitve applications, e.g., (soft) real-time applications. Otherwise TCP provides a more reliable data stream. MDD is intended for rather simply use cases. There are various libraries available for a more high-level data exchange, e.g., https://zeromq.org/ , but I'm not aware of any ports to Modelica language bindings. I've so far only used the OS functions directly.