modelica-3rdparty/Modelica_DeviceDrivers #337: Roadmap of extracting data in other software like MATLAB?

Satinelamp, 2021-04-08:
With the Modelica_DeviceDrivers library, we could send data to other software, but it seems there is no roadmap or instructions about how to extract data in other software like MATLAB.
Take the "Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP" as an example, there is a `uDPSend block`  and a `uDPReceive block`.
Here is the document of this example.
`The uDPSend block sends to the local port 10002. The uDPReceive block starts a background process that listens at port 10002. Consequently, the uDPReceive block receives what the uDPSend block sends. `
In order to extract the data, I need to know the data structure sent via UDP, but how should I extract data sent from Dymola in MATLAB? Is there a roadmap or instructions for this issue?
[image]

bernhard-thiele, 2021-04-08:
There is documentation within the MDD paper referenced in the project README.md, see Section "2.5.2  Communication" ([Download](https://www.modelica.org/events/modelica2017/proceedings/html/submissions/ecp17132713_ThieleBeutlichWaurichSjolundBellmann.pdf)). Please check if this answers your questions.

Satinelamp, 2021-04-08:
When sending strings, does MDD transfer the data with utf-8 encoding format? Does MDD support sending Japanese characters?

bernhard-thiele, 2021-04-09:
Mdd simply transfers the "const char*" byte buffer which it gets from the Modelica tool. There es no utf-8 encoding on MDD's side. However, if the Modelica tool writes the utf-8 into the byte buffer, it could work. I made a (very simple) test with Dymola and it seems that one can use utf-8 characters in strings. I've no experience with Japanese characters. Pasting in a random Japanese Hiragana unicode seemed to work in Dymola.

closed 2021-04-10 by Satinelamp