modelica-3rdparty/Modelica_DeviceDrivers #348: Serial port data unknown

max-privato, 2021-10-20:
I would like to use Modelica_DeviceDrivers to send data through a serial port.
I have two PCs connected through a serial cable and can transmit and receive text through a terminal app (I use Tera Term).

When I try modelica, instead, I receive unrecognised stuff.  I just  I send "Hello world!" through AddString (picture below). 

I think the first thing to check is the serial data: 
- data (7/8 bit)
- parity 
- stop bits
- flow control (Xon/Xoff, RTS/CTS, DSR/DTR, none)

While parity is under Modelica control, the other parameters are not. This is not bad per se. What is bad is that I don't know the values used by Modelica for them, and therefore cannot set the receiving PC to the right setting.

Can someone tell to me what values for the above communication parameters are used by SerialPortSend?

bernhard-thiele, 2021-11-16:
As mentioned in https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/issues/350#issuecomment-970150534 I'm not very knowledgeable in the serial port intricacies. I agree that the settings belong into the documentation and ideally should be the same on Windows and Linux :-). For the time being one needs to look into the C code external object constructors to figure out the configuration details (and I don't claim understanding everything in detail). In the external object constructor for Windows one finds

https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/blob/8267f35ef33f0893e65df5a3c047514936df6afd/Modelica_DeviceDrivers/Resources/Include/MDDSerialPort.h#L202-L212

and

https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/blob/8267f35ef33f0893e65df5a3c047514936df6afd/Modelica_DeviceDrivers/Resources/Include/MDDSerialPort.h#L221-L228

max-privato, 2021-11-18:
After I opened this ticket I successfully exchanged info between teraterm and OM via Modelica-DeviceDrivers+serial port successfully. To do this, I left Teraterm with the default setting, i.e. 8N1 (Parity none, o data bits, 1 stop bit)

So, this strongly suggests that this, 8N1,  is the setup of serial port of Modelica_DeviceDrivers.

Now I read on your message 
```
dcb.ByteSize = 8; 
dcb.StopBits = ONESTOPBIT; 
```
which confirms the assumption (moreover, I don't believe that the two ports can dialogue with unmatched settings)