modelica-3rdparty/Modelica_DeviceDrivers #351: Communication troubles after several seconds

max-privato, 2021-11-03:
I'm making "split simulations" with OM: one OM instance doing part of the simulation, the other the second part.
This is part of a project to realise Hardware-in-the-loop simulation taking advantage of OM. The two instances can work on the same PC, or on different PCs belonging to the same LAN.
The enclosed SplitSim.mo works without changes in two OM instances on the same PC

These split simulations are somehow frontier usages of OM, and if they work would enlarge OM capabilities significantly.
Everything works, except that on the instance running Slave_, the simulation stops at t=135.2 with the following message:

```
The code has encountered trouble from which it cannot recover.
can't continue. time = 135.200000
### STATISTICS ###
MDDUDPSocket.h: Receiving not possible, socket not valid.
DASKR-- TOUT (=R1) TOO CLOSE TO T (=R2) TO START INTEGRATION                    
      In above,  R1 =   1.3520000000000E+02   R2 =   1.3520000000000E+02
Simulation process failed. Exited with code 0xffffffffffffffde.
```
### Steps to Reproduce
The issue can be reproduced in a simplified way just running Slave::  the simulation stops at 135.2s with the error message above.

NOTES:
1. if one wants to see split simulation at work, they can run Master while Slave_ is already running.
2. Strangely, if I change the name of Slave_ into Slave its simulation does not start.

FIle (change extension to "mo" before using):
[SplitSim.txt](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/files/7464866/SplitSim.txt)

bernhard-thiele, 2021-11-16:
On first sight this seems rather an OM issue.

I've just tested with Dymola 2022 on Windows and it worked as expected.

However, for the Master model in Dymola I get following warnings:

```
Integration started at T = 0 using integration method DASSL
(DAE multi-step solver (dassl/dasslrt of Petzold modified by Dassault Systemes))
MDDRealtimeSynchronize.h: WARNING simTime=0.015916 < lastSimTime=0.025500 (possibly due to solver step-size adaption)
MDDRealtimeSynchronize.h: WARNING simTime=0.015916 < lastSimTime=0.025500 (possibly due to solver step-size adaption)
...
```

However, those go away if one uses the `sampled` option in the realtimeSynchronization block (which is already used in the Slave model).

For split simulation another option is to use the TCP/IP blocks. This allows the model with the TCP/IP server to control the step size progress of the model with a client. I have not experimented very much with this, but it seems to work rather nicely. In the tcpipserverconfig use `useNonBlockingMode=false`, in the `tcpipSend` block use `blockUntilConnected=false` and in the `tcpipReceive` block use `blockUntilConnected=true` and `showAdvancedOutputs=true`. When the server can be connected like this:

[image]

This gives you  control on the execution sequence of the input/output readings. So maybe an interesting alternative for using UDP, though the mutual blocking might be more costly in terms of (real-time) performance.

max-privato, 2021-11-21:
This opens new opportunities... and new issues.
To simulate Modelica models I use Dymola and OpenModelica. I prefer OpenModelica, for several reasons, the most important of which is that I prefer OpenSource software, also for my classroom teaching.
I tried your recommendation in the last comment, even though I'm not familiar with TCP/IP at all (but I can learn). Unfortunately, immediately I came across a new issue. I discussed it in this ticket:
https://github.com/OpenModelica/OpenModelica/issues/8172

bernhard-thiele, 2021-11-22:
Yes, I know this is an longstanding issue with OM and it is very nice if it gets fixed as indicated in your linked issue.

In hindsight to my previous post, I think I should comment a bit more about the caveats of using the advocated TCP/IP option. The example above would actually run the two models in serial. So it's not suitable for getting a speed-up for real-time applications.

For a speed-up in real-time simulation we probably would like to use UDP or shared memory. In addition it would be nice to have some sort of "director" or "master" to control the simulation federates, particularly if one tries to use more than two simulations. Some GUI for controlling the federates would be nice, too.

However, for the moment being, simply using UDP or shared memory and rely on the real-time clock for synchronization is probably the best one can do. More dedicated support for distributed real-time simulations would be cool, but would need some time to spare for implementing it properly...

closed 2023-04-13 by bernhard-thiele