modelica-3rdparty/Modelica_DeviceDrivers #364: MDDUDPSocket.h: socket failed with error: 10022

DaJansenGit, 2022-10-13:
I came across this error in some of our models using Modelica DeviceDrivers
```modelica
MDDUDPSocket.h: socket failed with error: 10022

The stack of functions is:
Modelica_DeviceDrivers.Communication.UDPSocket
Modelica_DeviceDrivers.Communication.UDPSocket(0, 0, true, 1, 0)
```
I found that the provided Example by Modelica DeviceDrivers throws the same error: `Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP`

I was also able to find the already merged PRs conversation in https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/pull/242 which had the same problem but as far as I understand was only solved for mingw32 setup.

The mentioned issue should be reproducable at least with the following setup:
* OS: Windows 10 64 bit
* Dymola 2023
* Compiler: Visual Studio Code 2019 and gcc 64 bit

bernhard-thiele, 2022-10-14:
I don't remember #242 very well, but the problem in the PR with the 10022 error was also fixed for the VS compiler. I can't reproduce the error with my setup (Dymola 2022 with VS 2017). Maybe its different with VS 2019. I should upgrade anyway and will test it.

DaJansenGit, 2022-10-14:
Thanks for the reply. Let me know if you can reproduce the error.

Thantalos, 2022-10-14:
Tried to reproduce the error with Win10/64bit + Dymola 2023 + Visual Studio 2019 + the current version of DeviceDrivers master branch.
```
Log-file of program ./dymosim
(generated: Fri Oct 14 11:16:43 2022)

dymosim started
... "Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP" simulating
... "dsin.txt" loading (dymosim input file)
MDDUDPSocket.h: Opened socket for sending.
MDDUDPSocket.h: Waiting for data on port 10012.
... "TestSerialPackager_UDP.mat" creating (simulation result file)

Integration started at T = 0 using integration method DASSL
(DAE multi-step solver (dassl/dasslrt of Petzold modified by Dassault Systemes))

Integration terminated successfully at T = 5
  WARNING: You have many time events. This is probably due to fast sampling.
  Enable logging of event in Simulation/Setup/Debug/Events during simulation
   CPU-time for integration                : 0.002 seconds
   CPU-time for one grid interval          : 0.004 milliseconds
   CPU-time for initialization             : 0.002 seconds
   Number of result points                 : 607
   Number of grid points                   : 501
   Number of accepted steps                : 445
   Number of f-evaluations (dynamics)      : 445
   Number of crossing function evaluations : 1155
   Number of Jacobian-evaluations          : 415
   Number of model time events             : 50
   Number of input time events             : 0
   Number of state events                  : 28
   Number of step events                   : 0
   Minimum integration stepsize            : 2.3e-07
   Maximum integration stepsize            : 0.483
   Maximum integration order               : 1
Calling terminal section
... "dsfinal.txt" creating (final states)
MDDUDPSocket.h: Receiving not possible, socket not valid.

SUCCESSFUL simulation of Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_UDP
```

Not so sure why I get `MDDUDPSocket.h: Receiving not possible, socket not valid.` but besides that everything seems to work just fine.

The error code isn't much help either: https://learn.microsoft.com/de-de/windows/win32/winsock/windows-sockets-error-codes-2#WSAEINVAL
Invalid argument - that could be anything.

bernhard-thiele, 2022-10-14:
Thanks @Thantalos for testing it for me (may take some time until I get a new VS installed).

This is the same log output that I get with VS 2017. The warning at the end of the simulation is probably because the socket is closed before the receiving thread has terminated. It should have no effect on the simulation results.

@DaJansenGit: Seems we can't reproduce the error. Did you try on another system as well, or maybe try after rebooting the machine?

tbeu, 2022-10-14:
> Not so sure why I get `MDDUDPSocket.h: Receiving not possible, socket not valid.` but besides that everything seems to work just fine.

That's already reported by #346.

DaJansenGit, 2022-10-26:
Sorry for taking that long to come back to you. I just tested it again at got the same error first. BUT was able to simulate successful after changing the working directory... 
For any reason, there seems to be a problem with the network UNC path that was configured by default.

Maybe this will help someone else in the future. Thanks for your help! I will close this as it's not DeviceDrivers related.

closed 2022-10-26 by DaJansenGit