modelica-3rdparty/Modelica_DeviceDrivers #370: Cosimulation using TCP/IP

AHaumer, 2023-04-01:
I want to establish a cosimulation between two modelica models running on two different computers. 
Using the TCP/IP-blocks from DeviceDrivers seems to be a good idea:
- The server starts and waits for the client to connect
- The client connects
- Loop
- - The client sends client’s values and waits for the server’s answer
- - The server receives the client’s values and sends the server’s answer

This way client and server should be synchronized.
Looking at the examples, I tried to implement a simple example consisting of a server and a client model. However, I get the error “singular” for both server and client. I'd be glad for any hints.
[CoSim.zip](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/files/11129971/CoSim.zip)

bernhard-thiele, 2023-04-04:
There seems to be a problem with the automatic sizing of connectors. Changing the "getters" from `.GetReal getReal(y(fixed=true), nu=1, n=1))` to `GetReal getReal(y(fixed=true))` compiles the models in Dymola 2022. I've made a few more changes (e.g., setting the parameter to the server that it waits until the client connects and made sure that message buffer sizes are minimal and equal in server and client) and the scenario above seems to work now.
[CoSim2.zip](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/files/11145611/CoSim2.zip)

AHaumer, 2023-04-05:
Thanks a lot @bernhard-thiele, just tried it with Dymola 2023x and it works!
What's the problem with the "getters"?
How did you set the server to wait until the client connects? Is it the "useNonBlockingMode=false"?

This offers some opportunities for me:
I'm able to co-simulate two Modelica models, running on different computers, linked via TCP-IP.
It's also possible to co-simulate a Modelica model with a python software (up to now we weren't able to export the Python software as fmu). Using TCP-IP (which should be possible) allows the two partners running on different computers.

bernhard-thiele, 2023-04-05:
@AHaumer happy to hear that it works. In addition to `useNonBlockingMode=false` in the `tcpipserverconfig` block I've set the following parameters for the server's receive block (`tcpipReceive`):

[image]

This gives the additional output `recvTrigger` which "Triggers if new data was received". By using this trigger for (eventually) triggering the execution of the `tcpipSend` block it is possible to enforce an order in the sequence of how the (external) functions are called. Here, we first would like to receive a package from the client and after that send back an answer to that package. The idea is shown in `Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_TCPIPServer`.

I think the problem with the "getters" and "setters" is that they were meant to be used very flexibly, but this has resulted in complicated logic. It is easy that something (like the automatic sizing) doesn't work as expected (maybe the blocks were copy-pasted form somewhere else and now the magic doesn't work) and the debug messages are only helpful if you have some experience with what can go wrong...

I guess I can close the ticket, otherwise feel free to reopen it.

closed 2023-04-05 by bernhard-thiele

AHaumer, 2023-04-06:
Thanks again @bernhard-thiele that helps a lot.
Of course ok to clode the ticket. 
Just a last question: 
The server is now configurated to first receive data, then send data.
The client has just one block. I suppose looking inside the client first sends data, then reveives data.
This way server and client are perfectly synchronized as necessary for cosimulation.

bernhard-thiele, 2023-04-12:
@AHaumer sorry for the late answer. I've been offline the past few days. Yes, that's how it should work.