modelica-3rdparty/Modelica_DeviceDrivers #371: TCP/IP with OpenModelica?

AHaumer, 2023-04-12:
I tried to execute my small example of interprocess-communication between two Modelica processes with OpenModelica.
It works fine with Dymola, but with OpenModelica but translation fails.
See enclosed zip with Modelica source and Error message from OpenModelica. 
Any hints how to get the example running with OpenModelica?
[CoSim_TCPIP.zip](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/files/11213725/CoSim_TCPIP.zip)

beutlich, 2023-04-12:
Should it rather be an issue for the OpenModelica issue tracker?

AHaumer, 2023-04-12:
Yes I forgot to mention: OpenModelica 1.20.0 on Windows 10 - sorry.
@beutlich I couldn't figure out whether it's an OpenModelica issue or something missing in DeviceDrivers.
@bernhard-thiele Yes your fix works, at least for the server. The client crashes:
The initialization finished successfully without homotopy method.
MDDTCPIPSocket.h: send failed with error code: 10053
MDDTCPIPSocket.h: recv failed with error code: 10053
MDDTCPIPSocket.h: MDD_SerialPackagerSetData failed. Buffer overflow.
Process crashed
Process crashed 
Simulation process failed. Exited with code 0xffffffffc0000005.

AHaumer, 2023-04-12:
I tried the following setup: 
Using the server process compiled by OM.
Starting the client process from Dymola 2023x.
Client crashes after the normal end (the client should terminate when he receives a Boolean true):
Error: The following error was detected at time: 5.02
MDDSerialPackager.h: MDD_SerialPackagerGetInteger failed. Buffer overflow.
The stack of functions is:
Modelica_DeviceDrivers.Packaging.SerialPackager_.getInteger
Modelica_DeviceDrivers.Blocks.Packaging.SerialPackager.Internal.DummyFunctions.getInteger
Modelica_DeviceDrivers.Blocks.Packaging.SerialPackager.Internal.DummyFunctions.getInteger(
getBoolean.pkgIn.pkg, 
1, 
getBoolean.pkgIn.dummy, 
getBoolean.byteOrder)
... "dsfinal.txt" creating (final states)
... Error message from dymosim
ERROR: The simulation of CoSim_TCPIP.Example3.TCPIP_Client FAILED

bernhard-thiele, 2023-04-13:
@AHaumer Nice to hear that the workaround works for you. I hope that somebody from OpenModelica finds some time for looking into the issue and can give some feedback about what is happening. If there is a simple "trick" which we can do in DeviceDrivers to make it work again, I would add it. However, I think it needs to be fixed in OpenModelica, so I'll close the issue here.

closed 2023-04-13 by bernhard-thiele