modelica-3rdparty/Modelica_DeviceDrivers #373: Struggling with serial port communication

tbeu, 2023-04-13:
@tbeu 
Dear Thomas
Yes it worked! We finally could test it and it worked perfectly, thank you so much : )

Now I am struggling with sending and receiving strings.
When I send a command e.g. "DL31" the board should e.g. turn the led on and then send back "DL31".

I am not able to receive strings in dymola from the board.
For my test example I use an **Arduino Uno** board (for now using the standard 8bit data) with this sketch:

```
void setup() {
  // Serial.begin(38400, SERIAL_7E1);
  Serial.begin(38400, SERIAL_8E1);
}

void loop() {

  Serial.write("Hallo\0");
  delay(1000);
}
```

and the following **modelica model**:

[image]

[image]

**Modelica code**:
```
model Test_ReceiveString_06
  extends Modelica.Icons.Example;

  Modelica_DeviceDrivers.Blocks.Communication.SerialPortReceive serialPortReceive(
    enableExternalTrigger=true,
    autoBufferSize=false,
    userBufferSize=6,
    Serial_Port="COM4",
    baud=Modelica_DeviceDrivers.Utilities.Types.SerialBaudRate.B38400,
    parity=1) "to send and receive data "
    annotation (Placement(transformation(extent={{-50,50},{-30,30}})));
  Modelica.Blocks.Logical.Not not1 annotation (Placement(transformation(
        extent={{-10,-10},{10,10}},
        rotation=180,
        origin={20,60})));
  Modelica.Blocks.Sources.BooleanPulse booleanPulse(period(displayUnit=
          "s") = 2)
               annotation (Placement(transformation(
        extent={{-10,-10},{10,10}},
        rotation=180,
        origin={60,60})));
  Modelica_DeviceDrivers.Utilities.TriggeredPrint triggeredPrint(message=
        getString.data)
    annotation (Placement(transformation(extent={{-50,72},{-30,92}})));
  GetString                                                        getString(
      dataBitSize=8, bufferSize=6)
    annotation (Placement(transformation(extent={{-10,-30},{10,-10}})));
equation 
  connect(triggeredPrint.trigger, not1.y) annotation (Line(points={{-40,76},
          {-40,60},{9,60}},                 color={255,0,255}));
  connect(booleanPulse.y, not1.u)
    annotation (Line(points={{49,60},{32,60}}, color={255,0,255}));
  connect(not1.y, serialPortReceive.trigger) annotation (Line(points={{9,60},
          {-40,60},{-40,52}}, color={255,0,255}));
  connect(serialPortReceive.pkgOut, getString.pkgIn)
    annotation (Line(points={{-29.2,40},{0,40},{0,-9.2}}, color={0,0,0}));
  annotation (Icon(coordinateSystem(preserveAspectRatio=false, extent={{-140,
            -120},{140,120}})),Diagram(coordinateSystem(preserveAspectRatio=
            false, extent={{-140,-120},{140,120}})),
    experiment(StopTime=30, __Dymola_Algorithm="Dassl"));
end Test_ReceiveString_06;

```

I get the following **error**:

[image]

I studied all the examples within the MDD Library but still not understanding how to make it work.
Must all strings be null terminated with `\0` or could they also be terminated with carriage return `\r` or new line feed `\n` and so on?
Could you help me with this too?
Should I create a new issue for this?

Vielen Dank und beste Grüsse!

_Originally posted by @sertho in https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/issues/368#issuecomment-1505978002_

tbeu, 2023-04-13:
When checking some questions ([SO](https://stackoverflow.com/q/10989364/8520615) or [Arduino.cc](https://forum.arduino.cc/t/serial-write-null-character/135210)), can you please try

```
Serial.write("Hallo\0", 6);
```
or
```
Serial.write("Hallo");
Serial.write((byte)0);
```

tbeu, 2023-04-13:
> Must all strings be null terminated with `\0` or could they also be terminated with carriage return `\r` or new line feed `\n` and so on?

Yes, `MDD_SerialPackagerGetString` requires a null-terminated buffer, see

https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/blob/8ac7971cf17aa4d8102dff30312266ff51ef428e/Modelica_DeviceDrivers/Resources/Include/MDDSerialPackager.h#L457-L495

sertho, 2023-04-14:
> When checking some questions ([SO](https://stackoverflow.com/q/10989364/8520615) or [Arduino.cc](https://forum.arduino.cc/t/serial-write-null-character/135210)), can you please try
> 
> ```
> Serial.write("Hallo\0", 6);
> ```
> 
> or
> 
> ```
> Serial.write("Hallo");
> Serial.write((byte)0);
> ```

Both suggestions seem to work : )
Even if I activate `autoBufferSize` within the `serialPortReceive` component.
I will go on on this and let you know if it works to build-up a serial communication using only strings.
Thank you!

sertho, 2023-04-14:
> Yes, `MDD_SerialPackagerGetString` requires a null-terminated buffer, see [...]

What if my machine's controler returns not null terminated but CR or LF terminated strings?

Could I then adopt the header file [Modelica_DeviceDrivers/Modelica_DeviceDrivers/Resources/Include/MDDSerialPackager.h](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/blob/8ac7971cf17aa4d8102dff30312266ff51ef428e/Modelica_DeviceDrivers/Resources/Include/MDDSerialPackager.h#L457-L495) ?

Is this also a header-only library?

tbeu, 2023-08-06:
@sertho Can this be closed by now?

sertho, 2023-08-07:
Please don't close it yet. I did'nt solve it by now.
Could you please help me with this:

> What if my machine's controler returns not null terminated but CR or LF terminated strings?
>
> Could I then adopt the header file [...]

Thank you!

tbeu, 2023-08-07:
@bernhard-thiele Actually, there is an inconsistency between `MDD_SerialPackagerAddString` and its counterpart `MDD_SerialPackagerGetString`: Whereas `MDD_SerialPackagerAddString` supports sending non-null-terminated strings (due to the used `memcpy` instead of `strncpy`), `MDD_SerialPackagerGetString` always expects null-terminated strings. What do you think of always adding a trailing null in `MDD_SerialPackagerAddString` in the sending buffer and in `MDD_SerialPackagerGetString` for the receiving buffer, if not found? I could create PR if you are fine with the proposed changes.

bernhard-thiele, 2023-08-07:
@tbeu Sounds good. Null-termination is granted for Modelica Strings, but this would make the API more robust if it is used from custom C code.

tbeu, 2023-08-07:
> What if my machine's controler returns not null terminated but CR or LF terminated strings?

PR #377 improves the output buffer handling by always adding a trailing null character instead of raising an error and terminating the simulation. You are then able to manually manipulate that string e.g. by means of the [Modelica.Utilities.Strings](https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_Utilities_Strings.html#Modelica.Utilities.Strings) functions.

sertho, 2023-10-26:
Hi

I could finally test your new implementation. But it does not work.
I am not sure if the error is on my side or is due to the implementation #377 .

For testing I downloaded the masterbranch yesterday.

I'm still not able to receive strings in dymola from the board.
For my test example I use an Arduino Uno board with the following sketch
(now using the new  byte size parameter for the 7bit data introduced with #378 ) :

```

void setup() {
  Serial.begin(38400, SERIAL_7E1); // using 7bit
  // Serial.begin(38400, SERIAL_8E1);
}

void loop() {
  
  // -------------------------------------------------------------------------------------------------------
  
  // -> this does not work
  // Dymola Output:
  // Warning: MDDSerialPackager.h: MDD_SerialPackagerGetString warning.
  // No terminating '\0' found in buffer of bufferSize: 4
  Serial.write("abc");
  
  // -------------------------------------------------------------------------------------------------------
  
  // -> this works, if in Modelica the parameter "getString.bufferSize" = 4 ("a", "b", "c" + "\0")
  // Serial.write("abc\0", 4);

  // -------------------------------------------------------------------------------------------------------
  
  // -> this works, if in Modelica the parameter "getString.bufferSize" = 4 ("a", "b", "c" + "(byte)0")
  // Serial.write("abc");
  // Serial.write((byte)0);
  
  // -------------------------------------------------------------------------------------------------------
  
  delay(1000);
}
```

and the following modelica model:

[image]

Modelica code:
```
model Test_ReceiveString_07
  extends Modelica.Icons.Example;

  Modelica_DeviceDrivers.Blocks.Communication.SerialPortReceive serialPortReceive(
    enableExternalTrigger=true,
    autoBufferSize=true,
    Serial_Port="COM6",
    baud=Modelica_DeviceDrivers.Utilities.Types.SerialBaudRate.B38400,
    parity=1,
    byteSize=7)
              "to send and receive data "
    annotation (Placement(transformation(extent={{-50,30},{-30,10}})));
  Modelica.Blocks.Sources.BooleanPulse booleanPulse(period(displayUnit="s")
       = 1, startTime=1)
               annotation (Placement(transformation(
        extent={{-10,-10},{10,10}},
        rotation=180,
        origin={60,40})));
  Modelica_DeviceDrivers.Utilities.TriggeredPrint triggeredPrint(message=
[... 133 lines cut ...]
      bufferSize,
      pkgIn.dummy);
  end when;

  annotation (preferredView="info",
          Icon(coordinateSystem(preserveAspectRatio=false, extent={{-100,
            -100},{100,100}}), graphics={Text(extent={{-150,136},{150,96}},
            textString="%name"),         Text(extent={{-150,82},{150,42}},
          textString="%Serial_Port"),            Text(extent={{-152,-48},{148,-88}},
          textString="%baud")}), Documentation(info="<html>
<h4>Support for receiving datagrams over a serial port</h4>
<h4>Example</h4>
<p>
See <a href=\"modelica://Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_SerialPort\"><code>TestSerialPackager_SerialPort</code></a>.
</p>
</html>"));
end SerialPortBidirectional_7bit;

```

Thank you for helping!
Let me know, if I should do other tests for debugging!

Beste Grüsse,
Thomas

tbeu, 2023-11-05:
> If I understand it right, this [#373 (comment)](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/issues/373#issuecomment-1667328587) means that it is not necessary anymore to add a null terminator from the board, so this should work, right?

Argh, an empty string buffer was returned for non null terminated string data. This is now addressed by #382.

PR #382 (2024-01-08): Accept non null terminated string on unpacking
This is a bug-fix for #378 and fixes #373.
closed 2024-01-08 by bernhard-thiele