modelica-3rdparty/Modelica_DeviceDrivers #383: CLOCK_MONOTONIC not available in -std=c89

maltelenz, 2023-11-08:
Here, `CLOCK_MONOTONIC` is used:

https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/blob/a59615399901dff407b7e9d359d7771351657c9d/Modelica_DeviceDrivers/Resources/Include/MDDRealtimeSynchronize.h#L637

This https://stackoverflow.com/a/8881705 says this is not available with the "proper" c standard version flags, and I indeed cannot compile the model in a tool which uses `-std=c89`:

```
Modelica_DeviceDrivers/Resources/Include/MDDRealtimeSynchronize.h:637:33: error: ‘CLOCK_MONOTONIC’ undeclared (first use in this function)
  637 |         int ret = clock_gettime(CLOCK_MONOTONIC, &rtSync->t_start);
      |                                 ^~~~~~~~~~~~~~~
```

Am I missing something, or is the library wrong?

bernhard-thiele, 2023-11-09:
Hm, for me this works in Dymola (openSUSE 15.4), even if I specify "-std=c89" in the compiler options. (At least after I remove a few comments  in `MDDRealtimeSynchronize` starting with `//` which are not accepted with `-std=c89`.)

What tool and which OS are you using?

maltelenz, 2023-11-09:
This was in a development version of System Modeler, and I see it across many platforms and compilers (Windows VS 2022, Linux GCC 8.5.0, Linux GCC 13.2, macOS Clang).

I can reproduce it stand-alone. This gives the error mentioned in my first message:

```
g++ -x c -std=c89 -m64 -fPIC -c Modelica_DeviceDrivers/Resources/Include/MDDRealtimeSynchronize.h
```

While if I change to `-std=gnu89` it compiles:
```
g++ -x c -std=gnu89 -m64 -fPIC -c Modelica_DeviceDrivers/Resources/Include/MDDRealtimeSynchronize.h
```

bernhard-thiele, 2023-11-09:
Thanks, I can replicate this. If we add `#define _POSIX_C_SOURCE 199309L` as proposed in your linked stackoverflow discussion, it compiles.

MDD assumes that it can directly call Windows system calls or Linux/POSIX system call (macOS is not supported). 

I'm not sure how harmful it is if I simply added the ``#define _POSIX_C_SOURCE 199309L` line to the code, like

```C
#elif defined(__linux__)

#define _POSIX_C_SOURCE 199309L
```
Would that solve that issue for you?

I would expect that you get a different error with VS 2022, since it #ifdefs to a different code?

maltelenz, 2023-11-09:
> Would that solve the issue for you?

I can confirm that it compiles stand-alone in this case, and it seems like a reasonable solution.

It doesn't solve the issue completely in System Modeler, but that is because of a tool issue, which we will work on.

> I would expect that you get a different error with VS 2022, since it #ifdefs to a different code?

Yes, apologies, that is a completely different issue, which I'll have to dig into a bit more. If it turns out to be a problem with the library, I'll open a separate issue.

bernhard-thiele, 2023-11-10:
Okay, should be fixed now.

closed 2023-11-10 by bernhard-thiele

bernhard-thiele, 2023-11-10:
@maltelenz Realized that my first "fix" introduces new problems, e.g.,

```
g++ -x c -std=gnu99 -m64 -fPIC -c Modelica_DeviceDrivers/Resources/Include/MDDRealtimeSynchronize.h
```

won't compile with `#define _POSIX_C_SOURCE 199309L`. After experimenting with gcc and clang I introduced some conditionals in [0f40a7b](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/commit/0f40a7bf6b66cd6adbf90ff57d8c3bcc6d79a17d). It is guess work, but I hope it is robust enough for most practical use cases.  I'm open for better proposals...

maltelenz, 2023-11-14:
> Realized that my first "fix" introduces new problems, e.g.,
> 
> ```
> g++ -x c -std=gnu99 -m64 -fPIC -c Modelica_DeviceDrivers/Resources/Include/MDDRealtimeSynchronize.h
> ```
> 
> won't compile with `#define _POSIX_C_SOURCE 199309L`.

I guess that should technically not happen in a strict Modelica environment, since the Modelica standard [specifies C89](https://specification.modelica.org/maint/3.6/functions.html#external-function-interface)...

> After experimenting with gcc and clang I introduced some conditionals in [0f40a7b](https://github.com/modelica-3rdparty/Modelica_DeviceDrivers/commit/0f40a7bf6b66cd6adbf90ff57d8c3bcc6d79a17d):

This still seems to work for me in the development version of System Modeler.

commit a233c197 (2023-11-10): Make CLOCK_MONOTONIC available in -std=c89 (fixes #383)