modelica-3rdparty/Modelica_DeviceDrivers #386: Support for OPCUA client

TeoGoddet, 2024-02-09:
Hello, 
Would you be interested to have an OPCUA client being contributed to this lib ?

tbeu, 2024-02-09:
This would be very interesting, but there might be some potential conflict of interests due to OPCUA library by DLR (being closed source and not part of M_DD): https://ecp.ep.liu.se/index.php/modelica/article/view/197/157

bernhard-thiele, 2024-02-09:
Well, we have been thinking of making the DLR OPC UA library open source, but there is no final decision, yet. It would be a distinct library (not integrated into M_DD), because the library is quite complex in itself (client and server functionality, including things like (Modelica) code generation from NodeSet2 files). I've been working on a new and more generic approach and API which finally should replace the API described in the paper. This new code already works for internal projects, but it's not complete and polished, so the library is not in a good state for releasing it right now without finishing the started stuff and cleaning up the legacy code.

Are you aware that there exists already a Modelica OPCUA-client library from Wolfram? (I think they have made it open source now, but haven't tried it, yet.)

Of course, I would be interested hearing about your ideas, the intended API, the feature set and the approach of realizing it. However, as @tbeu hinted, I'm doing work in that area and it is not open source (yet).

TeoGoddet, 2024-03-15:
The wolfram library is not open source, 
I have done a minimalist library based on open654 to interact with OPCUA

bernhard-thiele, 2024-03-18:
Yep, our library is also based on open62541. The OPC UA standard is extensive and from that perspective I think it makes more sense to have it as its own (Modelica) library instead of maintaining it as part of M_DD. Would be interesting to join efforts for such a library, but since our library is not (yet) an open-source library, it's unfortunately not possible (yet). I'll try to remember mentioning it here if there is a decision about the licensing of our internal library, but I don't expect that we will discuss this internally before June (because completing the revision and cleaning up the library will still take some time).