modelica-3rdparty/Modelica_DeviceDrivers #390: Can someone confirm whether the example "Modelica_DeviceDrivers.Blocks.Examples.TestSerialPackager_SerialPort" works correctly?

thomasspallek, 2025-01-04:
**Issue Description**
I've tried running the mentioned example in Dymola 2025x, but it seems that the sent data is neither received nor processed correctly. Specifically:
- The string "examp" does not appear to be found.
- The integerExpression sequence (0, 1, 2, 3, 4, 5, 4, 3, 2, 1, 0, 1, 2...) is not visible on the receiving side.

**Configuration Details**
- Sending via: Virtual COM2
- Receiving via: Virtual COM3

**Debugging Steps Taken**
To verify the functionality of the virtual ports, I removed the "Read part" and used PuTTY to monitor COM3. Based on my test, the bridged virtual ports (COM2 and COM3) seem to be functioning correctly, as COM3 successfully receives data sent from the Modelica model (see picture below).

[image]

I also tried sending "Hello World" every 2 seconds via an STM32 microcontroller (connected via USB on COM6). While the string was occasionally detected (findString), it was far from being received at 2-second intervals—instead, it appeared approximately every 100 seconds, depending on the parameter settings in the components.

tbeu, 2025-01-04:
Which version of M_DD did you use? Latest release or latests HEAD?

thomasspallek, 2025-01-04:
After further testing, I found that the example works perfectly on another machine. This confirms that the issue is not with the example itself but likely specific to my initial setup. Apologies for any inconvenience caused.

closed 2025-01-04 by thomasspallek